vtl/volvo-cem-cracker #4: Problem with cracking some PIN codes in P2

smyku22, 2021-02-02:
I noticed that with some PIN codes, for example: 179955542746, the algorithm is not able to crack it.

vtl, 2021-02-02:
Show the log?

smyku22, 2021-02-02:
[Crack_PIN_log.txt](https://github.com/vtl/volvo-cem-cracker/files/5914106/Crack_PIN_log.txt)

vtl, 2021-02-02:
Try to increase SAMPLES to 50 or 60. Or reduce CALC_BYTES to 2.

PrancingMoose, 2021-02-02:
Sorry to jump on here but how do you run the program, am using Windows 7 pro 64Bit, need to crack my cem pin, thanks

vtl, 2021-02-02:
You need to solder the Arduino (Teensy 4.0 + CAN-bus), compile the source code and program the hardware.

smyku22, 2021-02-02:
[Crack_PIN_100_samples_log.txt](https://github.com/vtl/volvo-cem-cracker/files/5914571/Crack_PIN_100_samples_log.txt)
I increased the number of samples to 100 and it gets worse. He has a problem with the second PIN byte.

vtl, 2021-02-02:
Your MCP2515 seems to add a lot of noise. The one I have gives a latency aligned to 2 microseconds (see video), while yours disperse the latency in every bucket around CEM's latency.

The code also supports Teensy's built-in CAN-controllers, you only need to add a CAN-transceiver, like SN65HVD230. Just attach its RX/TX pins to the Teensy's pins related to CAN.

I think I'll drop MCP2515 support, since it is so unreliable. Also a built-in simplifies the schematics, since it does not need a level shifter.

smyku22, 2021-02-02:
But as I don't use MCP2515. I use Teensy 4.0 and SN65HVD230. Could you please check this PIN on your CEM? If you want, I can send you the Flash program, from my CEM. I discovered that if the second byte of PIN is 99 then there is a problem with cracking. When the 2nd PIN byte changes to any other, the CEM is able to crack.

maxdenisov, 2021-02-25:
Hi, does anyone have success through ODB2? 
I have some issues with it:

pin[2] candidate: 81 with latency 709670
Candidate PIN 0x74 0x54 0x81 -- -- -- : brute forcing bytes 3 to 5 (3 bytes), will take up to 695 seconds
Progress: 0%..done

found PIN: 0x55 0x54 0x00 0x74 0x00 0x81
PIN is cracked in 1372.35 seconds
Validating PIN
PIN verification failed!
done

closed 2021-03-18 by vtl

frankhommers, 2023-05-21:
I got it working when brute forcing the last 4 pairs. The first 2 pairs were always the same result, after that it was basically random.