vtl/volvo-cem-cracker #15: 2002 volvo v70xc , Need to read the pincode from CEM.

vikingzvolvo, 2022-12-02:
Can anyone assist in this Teensy and Arduino thing ? how to go about it after ordering these two things?
All keys lost scenario. Autel does not cover the chip in CEM. And i want to be able to read from the obd port. Thanks !

vtl, 2022-12-02:
Brick-style CEMs (99-04) are not crackable yet. You need to take the CEM out and solder a bunch of wires to read it with BDM programmer. See MVS forum, it has an Arduino-based solution.

Or blow the flash chip off with hot air and read it normally in a programmer.

vikingzvolvo, 2022-12-12:
Hi, sorry for late reply but thanks for your email to me. I actually found
online and also saw on YouTube a video where someone actually accessed this
Motorola chip version on these cars right from OBD. And retrieved pincode !
 I am not able to get a reply from this YouTuber and I will find that video
again and forward you the link.
  It has been really done really achieved and I just want to get to know
exactly how !

If you like stay in touch maybe we may end up being friends or just solving
a situation like this. Thanks.

Imran khan

imransterling, 2023-01-18:
> Brick-style CEMs (99-04) are not crackable yet. You need to take the CEM out and solder a bunch of wires to read it with BDM programmer. See MVS forum, it has an Arduino-based solution.
> 
> Or blow the flash chip off with hot air and read it normally in a programmer.

Apparently you are not really aware of facts concerning this matter. Read it for yourself !
A research project grown out of curiosity. Cracks 6 bytes of pin code via High Speed CAN-bus in under 20 minutes.

Supported platforms:

P1:
2004 - 2011 S40
2004 - 2011 V50
2008 - 2013 C30
2006 - 2013 C70
P2:
2005 - 2006 S80
2005 - 2007 V70
2005 - 2007 XC70
2005 - 2009 S60
2003 - 2014 XC90
**### **Earlier P2 1999-2004 can be supported as well, CEM donation is welcome.****

Find us at Matthew's Volvo Site for support: https://www.matthewsvolvosite.com/forums/viewtopic.php?f=10&t=85611

RickHaleParker, 2023-01-21:
> I didn't understand what you've tried to say here.

Artifact in the Readme "Earlier P2 1999-2004 can be supported as well".

sanjurjo-miguel, 2024-03-04:
Hey everybody,
has there been any progress on this? Is "Brick-style CEMs (99-04) are not crackable yet" still current?
Is there any chance of these CEM being crackable soon?
Thanks,

vtl, 2024-03-05:
Nobody has spent any time on brick-shaped CEMs after that.

sanjurjo-miguel, 2024-03-06:
My V70 from 2004 has a brick-shaped CEM (30657629) that I need to be cracked.
For that purpose I would very much like to contribute to this project in any way: I'm fit in coding and know my way around electronics, embedded devices, etc... but I'm inexperienced in this particular field (CEM cracking) though... could you give me some hints on how to get started?

vtl, 2024-03-06:
> My V70 from 2004 has a brick-shaped CEM (30657629) that I need to be cracked. For that purpose I would very much like to contribute to this project in any way: I'm fit in coding and know my way [...]

Read CEM flash directly and find PIN in the file: https://www.matthewsvolvosite.com/forums/viewtopic.php?p=609357#p609357

We brick-shaped CEMs can't be cracked: https://www.matthewsvolvosite.com/forums/viewtopic.php?p=582635#p582635, and the next post.