vtl/volvo-cem-cracker #17: Volvo XC90 CEM pin crack pcb

Constantinos11, 2023-01-26:
Good evening to everyone
I have been for a cheap but effective way to unlock my cem pin.
I found this method but I'm having a hard time with a pcb file which I found for the job
The pcb has all the wiring, it just needs the teensy 4.0 and the SN65HVD230 transceivers.
However, on the schematic, I can see that some resistors that need installing, but I'm not sure where to install them on the pcb. 
I know that some resistors are only used for bench cem cracking so I should make it know that I intend to do the cem cracking through the obd2 port.
Another query I have is whether to get the teensy 4.0 with the pin already on or just the bare board

The pcb link:
https://www.pcbway.com/project/shareproject/Volvo_CEM_Pin_Decoder_c4d037d5.html

I look forward to your reply.

Best wishes,
Constantinos

RickHaleParker, 2023-01-27:
That is the PCB for the Bosh CF160 version. See: Bosh CF160 version.pdf in the docs folder.

Constantinos11, 2023-01-27:
I have seen it, but since I can’t find any Bosch cf160 in my country, I was wondering if I could use SN65HVD230 transceivers instead of the Bosch cf160.
Do the SN65HVD230 transceivers need receivers?

vtl, 2023-01-27:
No, transceiver does both RX and TX.

RickHaleParker, 2023-01-27:
A Ebay seller in Texas has a bunch of Bosch CF160. He will ship world wide.

Constantinos11, 2023-01-27:
Can you provide a link as I have not been able to source them

Constantinos11, 2023-01-27:
> No, transceiver does both RX and TX.

Does this mean that I can use this pcb with SN65HVD230 transceivers instead of Bosch CF160 and not add resistors?

vtl, 2023-01-27:
> > No, transceiver does both RX and TX.
> 
> Does this mean that I can use this pcb with SN65HVD230 transceivers instead of Bosch CF160 and not add resistors?

Yes. Original schematic uses them.

https://github.com/vtl/volvo-cem-cracker/blob/master/doc/schematic.png

Constantinos11, 2023-01-27:
> > > No, transceiver does both RX and TX.
> > 
> > 
> > Does this mean that I can use this pcb with SN65HVD230 transceivers instead of Bosch CF160 and not add resistors?
> 
> Yes. Original schematic uses them.
> 
> https://github.com/vtl/volvo-cem-cracker/blob/master/doc/schematic.png

Yes, it does, however the schematic includes two resistors (r1 and r2) which are not included on the pcb.

vtl, 2023-01-27:
> Yes, it does, however the schematic includes two resistors (r1 and r2) which are not included on the pcb.

I didn't do any PCB. SN65HVD230 datasheet asks for 10k resistors. 120 Ohm resistors depend on how the CEM is attached (on bench, in car). Buy any CAN shield with SN65 or Bosch, they will have resistors and jumper to disconnect 120 Ohm one.

Constantinos11, 2023-01-27:
>

> I didn't do any PCB. SN65HVD230 datasheet asks for 10k resistors. 120 Ohm resistors depend on how the CEM is attached (on bench, in car). Buy any CAN shield with SN65 or Bosch, they will have [...]

I meant the pcb that I had linked

https://www.pcbway.com/project/shareproject/Volvo_CEM_Pin_Decoder_c4d037d5.html

However, I can see that this would only work with cf160 because the SN65HVD230 needs 10k resistors and I don’t know where in the pcb to add them.

Would you recommend using the pcb or just making it by using a breadboard instead?

RickHaleParker, 2023-01-28:
> Can you provide a link as I have not been able to source them

https://www.ebay.com/itm/192811194444?hash=item2ce471584c:g:dBIAAOSw1NNg-vj6&amdata=enc%3AAQAHAAAAoEQ3HRoegrPH1YvnTj5VdUantK%2Ftkn%2F2MfUg9PYJEAmcE7jI5pomgMe9o2F2Dov82m3VjwCdDE9T243zNnukLkvqU5pY552Wl97sAQIJHqW2lvJaJ9xEO4mhMY%2FdgKuccDDxiVL7L4hrZwGETKT3PImbmAUO1TZe7UiUWV6pUgrio8vsrgYoZeyEAnAfLfeC7SsrUtn5g6Eh9b%2FS3o4nMBk%3D%7Ctkp%3ABk9SR_TXhue-YQ

RickHaleParker, 2023-01-28:
> I meant the pcb that I had linked
> 
> https://www.pcbway.com/project/shareproject/Volvo_CEM_Pin_Decoder_c4d037d5.html
> 

That PCB is for this build version:
https://github.com/vtl/volvo-cem-cracker/blob/master/doc/Volvo%20CEM%20PIN%20Cracker.%20Bosh%20CF160%20version.pdf

Constantinos11, 2023-01-28:
Would TCAN1042HVDR work?

vtl, 2023-01-28:
> Would TCAN1042HVDR work?

Perhaps

Constantinos11, 2023-01-29:
Do you know how I can verify this before buying all the parts?

RickHaleParker, 2023-01-29:
> Would TCAN1042HVDR work?
That V in the suffix makes it questionable. What would the I/O voltage be with Vio floating? 
A version like TCAN1044A-Q1 ( no V in suffix ) looks like it would work. 

Minimum order on the PCBs is 5. You would have some extra PCBs.

vtl, 2023-01-29:
> Do you know how I can verify this before buying all the parts?

Compare datasheets. I did, and see nothing major against using it. VCC to +5v, VIO to 3.3v.

RickHaleParker, 2023-02-05:
> > Do you know how I can verify this before buying all the parts?
> 
> Compare datasheets. I did, and see nothing major against using it. VCC to +5v, VIO to 3.3v.

Constantinos11, 2023-04-20:
https://github.com/vtl/volvo-cem-cracker/blob/master/doc/schematic.png
In this pcb, where does the CAN-GND (OBD-II pin 5) connect to apart from the obd plug?

vtl, 2023-04-20:
> https://github.com/vtl/volvo-cem-cracker/blob/master/doc/schematic.png In this pcb, where does the CAN-GND (OBD-II pin 5) connect to apart from the obd plug?

All ground points are connected together.