vtl/volvo-cem-cracker #25: Unknown CEM error

Drekith-1, 2024-03-27:
Hi. New here first off I’m not a programmer or anything just a fairly tech savvy handsy guy with too much free time. Anyway I’ve followed the schematics for the lcd version using a teensy 4.0 
However when I try to decode the cem the lcd displays unknown cem exiting……..

I’ve triple checked my connections are quality and true to the schematic as well as reloading the software onto the teensy a few times yet I get the same error. Is it likely I’ve just done something wrong or is it just not possible with my cem. 

V50 2007/2008 D5 facelift

vtl, 2024-03-29:
https://github.com/cmolson/volvo-cem-cracker/tree/one_pass

Drekith-1, 2024-03-31:
Worked a treat. Another successful read thanks guys 👌 drls are finally now off via vdash. 
[image]

closed 2024-03-31 by Drekith-1